Juniper/open-register-design-tool #7: Generated systemVerilog RTL for W1S bit can get stuck if hwclr event coincides with write event of different bit  in same register.

tkafafi, 2017-04-27:
Consider following rdl:
field REQ  { reset = 0; hw = r;  sw = rw; hwclr; woset;  }; //RW1S
addrmap  {   regfile Registers  {
// Request Register
            reg {
               REQ fld5[1:1];
               REQ fld6[8:8];
               REQ fld7[0:0];
            } d_reg;
    } regs;} foo;

Consider following stimulus:
- write d_reg = 0x002; // set fld5 request bit;
- write d_reg = 0x100; // set fld6 request bit
- pulse fl5_hwclr bit for one cycle at same time that d2l_bar_d_req_we (in pio_logic.sv) is high.
- final result is fld5 request it will not clear and will remain set, even though hw issued the hwclr event for it.

The root cause the is in pio_logic.sv, the hwclr priority should be higher (not lower) than the write priority.
Attached is simulation waves of the bug and the proposed fix.

Thanks
[image]

[image]

sdnellen, 2017-04-27:
Fix for default handling of woset/woclr checked in 170427.01 - will be picked up in next release.  For now should be able to override precedence setting to get correct behavior, eg:
field REQ { reset = 0; hw = r; sw = rw; hwclr; woset; precedence=hw; }; //RW1S

tkafafi, 2017-04-28:
Thanks, the precedence=hw worked.

sdnellen, 2017-05-04:
Fix in 170504.01

closed 2017-05-04 by sdnellen