Juniper/open-register-design-tool #8: Generated reglist text file missing the field descriptions (even though show_field is set true)

tkafafi, 2017-04-27:
Consider the following RDL:
field CFG  { reset = 0; hw = r;  sw = rw; }; //RW
regfile Registers  {
            reg {CFG fld1[9:0];CFG fld4[16:16];} a_reg;
};
addrmap {
Registers regs;
regs.a_reg.fld1->desc="Description of field1";
regs.a_reg.fld4->desc="Description of field4";
} foo;

Consider following setup in test.parms:
// reglist output parameters
output reglist {
        display_external_regs = true       // include external regs in output?
        show_reg_type = false              // show int/ext type for each reg?
        show_fields = true                // show field info for each reg
}

Generated Reglist file is below, and doesn't have field descriptions.:
(
//   Ordt 170308.01 autogenerated file
//   Input: top.rdl
//   Parms: test.parms
//   Date: Wed Apr 26 15:27:51 PDT 2017
//
0x0              size: 32   fields: 2    bits: 11        regs.a_reg
  fld1 [9:0]                 mode: RW   reset: 0                     ( regs.a_reg.fld1 = 0 )
  fld4 [16]                  mode: RW   reset: 0                     ( regs.a_reg.fld4 = 0 )

Instead, output should be:
0x0              size: 32   fields: 2    bits: 11        regs.a_reg
  fld1 [9:0]                 mode: RW   reset: 0                     ( Description of field1 )
  fld4 [16]                  mode: RW   reset: 0                     ( Description of field4 )

 Thanks

sdnellen, 2017-04-27:
Behavior is by design.  The show_fields parameter adds only basic field info to the file (name, path, access mode, size, etc).  To keep reglist file size manageable, descriptions of registers and fields are not included.  If more detailed info is required (eg for post processing), suggest using the xml output format.

tkafafi, 2017-04-28:
Ok, thanks. Will post-process xml as you suggested.
Thanks

closed 2017-05-04 by sdnellen