Juniper/open-register-design-tool #9: Feature Request: Support asynchronous resets in generated SystemVerilog

tkafafi, 2017-04-27:
Generated System Verilog RTL already supports synchronous resets for all sequential elements. The change to support asynchronous resets is very minor.
Basically the sensitivity of always blocks would need to add posedge reset, as below
Synchronous reset always block:
  always_ff @ (posedge clk) begin
    if (reset)  ......
    else ....
  end

Changes as below:
  always_ff @ (posedge clk **or posedge reset**) begin
    if (reset)  ......
    else ....
  end

Thanks

sdnellen, 2017-04-28:
Added use_async_resets boolean systemverilog output parameter to allow async resets 170427 - will be picked up in next release.

tkafafi, 2017-04-28:
Thanks, I appreciate it

sdnellen, 2017-05-04:
Feature in 170504.01 release

closed 2017-05-04 by sdnellen