Juniper/open-register-design-tool #10: Feature Request: support 'include directive to allow for hierachical RDL files

tkafafi, 2017-05-03:
Seems ORDT is expecting all chip registers to be in a single (flat) RDL file; while most chips would have hierarchical structure and thus a hierarchical RDL to match.

Most chips would generally be hierarchical which each block having a separate designer and he/she would right a separate block level RDL file. Then there would be RDL file at the subsystem level that would include (reference) the relevant block level RDLs. Then finally a top level RDL that would include (reference) the subsystem RDL.

There would also need to be an include_directory list in the parms file to tell ordt where (which directories) to look for the included files (similar idea to +incdir switch that simulators use) 

Thanks

sdnellen, 2017-05-04:
Typically file inclusion (as well as variable use/macro definition) would be done by preprocessing the source files with any number of available preproc utilities (SystemRDL 1.0 spec used a perl preprocessor in addition to vpp, but syntax is a bit flaky, so these are not implemented in ordt).  We use an inhouse perl routine to preprocess files here, which works quite well for large projects. 

Baseline would be to use the C or verilog preprocessors, but here's pointers to a few others I found that might be of interest (have not actually tried any of these):

gcc - https://logological.org/gpp
xpp - http://www.cross-browser.com/x/docs/xpp_reference.php
vbpp - https://github.com/balanx/vbpp
perlpp - https://github.com/MegabytePhreak/rdl/tree/master/perlpp
ep3 - http://search.cpan.org/~mykl/Text-EP3-1.10/lib/Text/EP3.pm
preprocess - https://pypi.python.org/pypi/preprocess/1.1.0

tkafafi, 2017-05-06:
Thanks for your suggestions. 
I found a simple preprocessor at: http://www.beyond-circuits.com/files/vpp.pl and its working well to process `include and `define
No installation is needed, just download the file, change the perl location on first line and call the script directly.

Thanks

closed 2017-05-31 by sdnellen