Juniper/open-register-design-tool #11: Feature Request: Support regwidth < 32

paulthilking, 2017-05-26:
The SystemRDL v1.0 spec says:
All registers shall have a width = 2N, where N >=3

The current ordt implementation reports an error when specifying regwidth = 16 for a register:

*** ERROR ***: Invalid register width (16) specified in xyz_reg_file.abc_reg.  Size should be between 32 and 1024.

sdnellen, 2017-05-31:
Basic support for registers less than 32b was committed 5/29 and will be picked up in the next release.  Note that use of <32b regs will require the global parameter *min_data_size* to be set to 8 or 16 (it defaults to 32).  This parameter controls the range of registers allowed (will be min_data_size to 16*min_data_size), optimization of lower address bits, and word definition in generated rtl.  Basic functionality of 16b registers was verified, but there are known issues in generation of serial8 and ring\* decoder interface forms with use of non-32b base.  It is TBD how much of this will be flushed out before next release.

sdnellen, 2017-07-11:
Fix picked up in release 170710.03.

closed 2017-07-11 by sdnellen