Juniper/open-register-design-tool #13: Address calculation issue

erosh, 2017-07-19:
Hello again.
Running ordt with attached file (renamed to *rdl) returns me an error:
"out of order register set address specified in me2"
But I think it shouldn't
[t.txt](https://github.com/Juniper/open-register-design-tool/files/1159813/t.txt)

One more - I'm getting:
*** ERROR ***: Replicated address maps are not allowed (map name = pf_slot) for
addrmap pf_slot
        {
         ...
         };
pf_slot     pf_slot[16]     @0x0000 += 0x0010;
How can I avoid it?

Thanks

sdnellen, 2017-07-19:
Nested addrmaps become cascaded decoders in rdl output w each having addl alignment constraints.  In most cases, use a single addrmap at top level and use regfile to group register underneath - regfiles (and addrmaps) should still be aligned to next greater power of 2 (by default).  Regfiles can be replicated also.  So, something like this...

addrmap top
{
	regfile me /* <--- SDN changed */
	{
	    regfile ME_MSK_MEM  /* <--- SDN changed */
	    {
	        reg mem
	        {
	            field
	            {
	                hw      = r;
	                sw      = rw;
	                desc    = "Memory cell";
	            } mem[31:0] = 32'h0;
	        };
	
	        mem     mem[64] @0x0000;
	    };
	    ME_MSK_MEM            ME_MSK_MEM            @0x0200;
	};
	me          me1               @0x0400;
	me          me2               @0x0800;  /* <--- SDN changed */
} ot;

erosh, 2017-07-20:
In our case cascaded decoders compare full addresses not only some bits, so they doesn't have such restriction (power of 2). So in my fork I'd like to change 
```
diff --git a/src/ordt/output/OutputBuilder.java b/src/ordt/output/OutputBuilder.java
index 61eed14..8163fcd 100644
--- a/src/ordt/output/OutputBuilder.java
+++ b/src/ordt/output/OutputBuilder.java
@@ -354,14 +354,14 @@ public abstract class OutputBuilder implements OutputWriterIntf{
 
        /** add a set of external registers to this output 
         * @param newRegProperties - if non-null (is ext regset, not reg), this will be set as external and used as static regProperties for output gen */
-       public void addRootExternalRegisters(RegProperties newRegProperties) { 
-               int reservedRange = updateRootExternalRegProperties(newRegProperties, false);
+       public void addRootExternalRegisters(RegProperties newRegProperties) {
+               RegNumber reservedRange = updateRootExternalRegProperties(newRegProperties, false);
 
                addRootExternalRegisters();  // note getExternalRegBytes() is usable by child
 
                // now bump the running address count by the reserved range
                RegNumber newNext = new RegNumber(getExternalBaseAddress());
-               newNext.add(new RegNumber(reservedRange * getMinRegByteWidth()));
+               newNext.add(reservedRange);
                setNextAddress(newNext);   
                //System.out.println("addRootExternalRegisters   base=" + getExternalBaseAddress() + ", next=" + getNextAddress() + ", delta=" + extSize);
        }
@@ -385,7 +385,7 @@ public abstract class OutputBuilder implements OutputWriterIntf{
         * @param newRegProperties - properties that will be updated
         * @param isNonRootExternal - if true, base address will be set to that of current regset else builder ext base addr will be used
         */
-       private int updateRootExternalRegProperties(RegProperties newRegProperties, boolean isNonRootExternal) {
+       private RegNumber updateRootExternalRegProperties(RegProperties newRegProperties, boolean isNonRootExternal) {
                // if new regProperties, then init 
                if (newRegProperties != null) {
                        if (!newRegProperties.isExternal()) {
@@ -404,16 +404,7 @@ public abstract class OutputBuilder implements OutputWriterIntf{
 
                // compute size of the external group from next and stored base address
                RegNumber extSize = getExternalRegBytes();  
-               //System.out.println("OutputBuilder addRootExternalRegisters: base=" + getExternalBaseAddress() + ", next=" + getNextAddress() + ", delta=" + extSize);
-
-               int lowBit = getAddressLowBit();  // same low bit as overall address range
-               regProperties.setExtLowBit(lowBit);  // save the low bit in external address
-               //int highBit = getAddressHighBit(repSize);
-               int range = getAddressWidth(extSize);
-               regProperties.setExtAddressWidth(range);  // set width of the external address for this group
-               int reservedRange = 1 << range;  // calc 2^n
-               //System.out.println("addRootExternalRegisters   ext addr range=" + range + ", lo bit=" + lowBit + ", new rep count=" + reservedRange);
-               return reservedRange;
+               return  extSize;
        }
 
        /** add a register set to this output 
```
I need only export in xml and html, so I hope it will work.

sdnellen, 2017-07-26:
Power of 2 constraint on cascaded address maps and external regions is due a limitation of the generated decoder - constraint allows a single address match to select the child region.  There are other alignment constraints for regsets enabled by default, most of which can be disabled by setting **use_js_address_alignment** to false, but the addrmap constraint will remain.  If only xml is being generated, you should not need to use cascaded addrmaps - these only affect physical structure in generated rtl.

closed 2017-07-31 by erosh