Juniper/open-register-design-tool #14: Add hardware-specific tags to XML output

kduncklee, 2017-07-19:
The current XML output focuses on a software view of the register space. It would be helpful for hardware generation if there was an option to output additional hardware-specific information.

This would include tags like:
- hw access type (RW, R, W, NA)
- saturate for counters
- we/wel
- enums

Should there be a param in the params file to control whether to output these fields or can we just let downstream tools ignore the tags they don't use?

sdnellen, 2017-07-20:
Checked in a fix in 170718.01 that adds **include_field_hw_info** parameter which adds most info used for hw gen - adds a hwinfo tag that contains general hw stuff (hwaccess, etc) and puts counter/interrupt info into their corresponding elements.  Enums are still not included, but would be a good addition.
Parm usage...

      // xml output parameters
      output xml {
      	include_field_hw_info = true   // include field hw characteristics 
      }

sdnellen, 2017-07-27:
Made **include_field_hw_info** true by default in 170727.01.

closed 2017-07-27 by sdnellen