Juniper/open-register-design-tool #15: Feature request: Add support for byte enables

ebertland, 2017-07-20:
We would like to have support for byte enables on the parallel interface and the internal register logic.

Is there a way to do this now? I could not find it in the documentation.

sdnellen, 2017-07-20:
No, byte enables are not supported currently.

sdnellen, 2017-07-27:
Added **write_enable_size** systemverilog output parameter in 170727.01 which will be picked up next release.  If non-zero, write enable input vector w/ each bit controlling **write_enable_size** bits in the data will be generated - so for byte enables, set to 8.  Internal registers use the enables directly and the enable vector is passed on to external parallel and sram regions.  No support in ring, serial, and leaf interface types and uvm model knows nothing of enable state.    See rdl_write_enable example in test area.

ebertland, 2017-09-10:
Thank you. I haven't had a chance to try this yet. Is there an automatic nightly build?

sdnellen, 2017-09-12:
Uploaded pre-release version 170911.01 for eval of new features.

sdnellen, 2017-09-15:
feature is in 170915.01.

closed 2017-09-15 by sdnellen