Juniper/open-register-design-tool #17: Feature Request: address validation and default value calculation

erosh, 2017-07-31:
It might be useful to add address validation to your rdl parser.
If I have field
```
field
{
	hw      = na;
	sw      = r;
	desc    = "Reserved";
} rsrvd[1:31] = 31'h0;
```
I will eat it.
As I understand it can be fixed just by adding one line inside RdlModelExtractor.

Is it possible now to calculate default value for addrmap based on registers default values?

sdnellen, 2017-07-31:
Added check for invalid field/fieldstruct bit ranges/widths in 170731.01.

Not clear what is being requested by 'calculate default value for addrmap based on registers default values'.  Can you provide example?

erosh, 2017-08-01:
Sorry. Default value of rigister based on field values
```

reg DPKT_CTRL_UNIVERSAL
    {       
        field
        {
            hw      = na;
            sw      = r;
            desc    = "Reserved";
        } rsrvd2[31:16] = 16'h0;

        field
        {
            hw      = r;
            sw      = rw;
            desc    = "Packet header length";
        } pkt_hdr_len[15:11] = 5'd16;

        field
        {
            hw      = r;
            sw      = rw;
            desc    ="";
        } mode[10:0] = 11'h0;
    };

```
so default value of this register should be 0x00008000

sdnellen, 2017-08-01:
Yes, generated code for register above will read 0x00008000 after reset.

erosh, 2017-08-02:
?
But how to get this value f.e. when converting into xml?
I can see only "reset" values for fields, but not for whole register

sdnellen, 2017-08-04:
Register level reset value can be calculated from its field reset/idx/width info in xml.  Would also likely generate a reset_valid vector as some fields may not have reset.

sdnellen, 2017-09-15:
bit range check is in 170915.01.

closed 2017-09-15 by sdnellen