Juniper/open-register-design-tool #18: Grammar error on anonymous definition of an external register

kduncklee, 2017-08-01:
The ordt parser throws an error on an anonymous definition of an external register. It works fine if the register type is defined and then instantiated.

Failing testcase:
```
addrmap {
  external reg {} test_reg[32];
} test_map;
```

Tool output:
```
line 2:11 mismatched input 'reg' expecting {ID, INST_ID}
line 3:0 extraneous input '}' expecting {<EOF>, '<PARMS>', 'property', 'default', 'signal', 'addrmap', 'reg', 'regfile', 'field', 'fieldstruct', 'internal', 'alias', 'external_decode', 'external', 'enum', 'arbiter', 'sharedextbus', 'errextbus', 'littleendian', 'bigendian', 'rsvdset', 'rsvdsetX', 'bridge', 'shared', 'msb0', 'lsb0', 'sync', 'async', 'alignment', 'accesswidth', 'addressing', 'clock', 'hwenable', 'hwmask', 'hw', 'sw', 'posedge', 'negedge', 'bothedge', 'level', 'nonsticky', 'name', 'desc', 'rset', 'rclr', 'woclr', 'woset', 'we', 'wel', 'swwe', 'swwel', 'hwset', 'hwclr', 'swmod', 'swacc', 'sticky', 'stickybit', 'intr', 'anded', 'ored', 'xored', 'counter', 'overflow', 'reset', 'cpuif_reset', 'field_reset', 'activehigh', 'activelow', 'singlepulse', 'underflow', 'incr', 'decr', 'incrwidth', 'decrwidth', 'incrvalue', 'decrvalue', 'saturate', 'incrsaturate', 'decrsaturate', 'threshold', 'incrthreshold', 'decrthreshold', 'dontcompare', 'donttest', 'regwidth', 'fieldwidth', 'signalwidth', 'precedence', 'encode', 'resetsignal', 'mask', 'enable', 'haltmask', 'haltenable', 'halt', 'next', 'nextposedge', 'nextnegedge', 'maskintrbits', 'satoutput', 'category', 'sub_category', 'js_attributes', 'js_superset_check', 'js_macro_name', 'js_macro_mode', 'js_namespace', 'fieldstructwidth', 'rtl_coverage', 'uvmreg_is_mem', 'uvmreg_prune', 'use_new_interface', 'use_interface', 'use_new_struct', 'use_struct', 'cppmod_prune', ID, 'XPROPERTYX', INST_ID}
```

sdnellen, 2017-08-01:
Rdl syntax is a bit flaky wrt/ externals.  Try

    addrmap {
      reg {} external test_reg[32];
    } test_map;

External modifier applies to the instance, not the component def.

kduncklee, 2017-08-01:
I was following the Accellera SystemRDL 1.0 spec. From 8.3.1.b on page 41:
> An anonymous definition (and instantiation) of an external register appears as follows.
> `external reg {[reg_body;]*} reg_instance [number];`

Defining the reg_name and then instantiating external is a fine workaround.

sdnellen, 2017-08-01:
Yup, their spec is inconsistent - antlr code in annex b of same spec (p77) doesn't allow external before the comp def.  This grammar was basis for ordt.

kduncklee, 2017-08-01:
Thanks, I'll update the RDL.

closed 2017-08-01 by kduncklee