Juniper/open-register-design-tool #21: Feature Request: Avoid the root addrmap instantiation or allow internal addrmaps instantiated

erosh, 2017-08-09:
F.e. I have 
A.rdl:
`addrmap A
{
	reg...
}a;
`
and B.rdl:
`addrmap B
{
'include "A.rdl"
...
}b;
`
after pre-processor I will receive instantiated addrmap A inside addrmap B. And its processing returns error;

sdnellen, 2017-08-09:
Pls send detail of error seen, rdl input after post-processing, and ordt command options being executed.

erosh, 2017-08-10:
$ java -jar ordl.jar -xml /samples/B_ext.xml /samples/B_preprocessed.rdl
Open Register Design Tool, version=170804.02, input=/samples/B_preprocessed.rdl
*** WARNING ***: No parameters file specified.  Default or inline defined parameters will be used.
Ordt: building xml...
*** ERROR ***: out of order register set address specified in A
Ordt exited due to error Thu Aug 10 16:37:38 MSK 2017

[B_preprocessed.txt](https://github.com/Juniper/open-register-design-tool/files/1215089/B_preprocessed.txt)
[A.txt](https://github.com/Juniper/open-register-design-tool/files/1215095/A.txt)
[B.txt](https://github.com/Juniper/open-register-design-tool/files/1215096/B.txt)

Btw I've got an error when set hw or sw flags with uppercase values f/e "R" or "RW"

sdnellen, 2017-08-10:
Looks like you are instancing addrmap A twice: once at the end on A.txt with name "a" and implicit address of 0x0 and then in B.txt with name "A" and explicit address 0x0.  This causes the out of order address error (since both are trying to occupy same range).  Change last line of A.txt from "}a;" to "};" should fix.

erosh, 2017-08-10:
yes. I know that. But if I want to convert A.txt(rdl) into *.xml I have instanciate it again. 
I mean A.txt -> .xml should have }a;
B.txt->.xml should remove 'a'.

sdnellen, 2017-08-10:
OK.  You'll need to handle a case such as this in the pre-processor using a variable (eg a DEFINE A_INST_NAME in A.txt or something similar).

closed 2017-09-14 by sdnellen