Juniper/open-register-design-tool #22: Feature Request: C header file output

ebertland, 2017-09-05:
We would like to use ordt to generate C header files from our SystemRDL files for people who are writing C and C++ drivers for our hardware.

We would like to have defines for each address offset (and struct style too), field msb and lsb and width at least. I can provide test-cases (input and expected output), if somebody is willing to work on this.

sdnellen, 2017-09-05:
Yes, headers are std way of handling reg info in drivers - send test-cases and will put it in queue.  Also recommend looking at cppdrvmod output option if your drivers are C++ - does the conversion from path to address and extracts field info, and has has ability to handle incremental changes between chip revs or reused reg spaces.

sdnellen, 2019-07-25:
Kamino cloned this issue to [sdnellen/open-register-design-tool](https://github.com/sdnellen/open-register-design-tool/issues/1)