Juniper/open-register-design-tool #26: Register Field Cannot Have Signal Assigned to Reset Property

ryansiglabs, 2017-09-16:
[test.sv.txt](https://github.com/Juniper/open-register-design-tool/files/1308728/test.sv.txt)

[test.rdl.txt](https://github.com/Juniper/open-register-design-tool/files/1308727/test.rdl.txt)
I am unable to use an externally supplied signal for the reset value of a register field.  The attached files re-create the problem.

The SystemRDL was copied from Section 6.3.2 of the SystemRDL v1.0 spec from Accellera.

ordt includes the signal as an input in the _pio and _logic modules, but it doesn't do anything with it.

The following is the console output:

C:\FPGA\graviton_sdr>java -jar Ordt.jar -systemverilog test.sv test.rdl
Open Register Design Tool, version=170915.01, input=test.rdl
*** WARNING ***: No parameters file specified.  Default or inline defined parameters will be used.
Ordt: building systemverilog...
*** WARNING ***: field reg1.a has no reset defined
*** ERROR ***: user defined signal sig_mySig is not used
Ordt: writing systemverilog file test.sv...
Ordt complete Sat Sep 16 16:13:58 PDT 2017

sdnellen, 2017-09-18:
fix added in 170917.01.   will be picked up in next release.

sdnellen, 2017-11-02:
Fix in 171102.01 rls

closed 2017-11-02 by sdnellen