Juniper/open-register-design-tool #27: SWMOD Property Produces Pulse Too Early

ryansiglabs, 2017-09-18:
Setting the swmod property of a field to true produces a one clock pulse on the appropriate signal out of the _pio module, but that pulse happens prior to the signal reflecting the fields new value getting updated.

This may be the intended behavior, but it seems incorrect to me at least.  The attached files can be used to re-create the issue in simulation.  Of note are the l2h_rf1_r1_f1_swmod_o and l2h_rf1_r1_f1_r signals.

ORDT version: 170915.01
[test.zip](https://github.com/Juniper/open-register-design-tool/files/1311858/test.zip)

Thank you,

Ryan

sdnellen, 2017-09-22:
Issue with changing this as there's a bunch of rdl code out there using swmod to qualify field->next in signal assigns.  This would break if we delay swmod. Could consider delaying only the actual output (prob ok) or adding a switch to control (not optimal).

ryansiglabs, 2017-09-23:
I'm not quite sure what is meant by "Could consider delaying only the actual output (prob ok)", could you elaborate?

I'm fine with registering the swmod_o signal out of the _pio module and using that to trigger logic that acts on changes to the field value reflected on the _r signal out of the _pio module.  If swmod_o is behaving as intended and ORDT isn't planning on changing its behavior in a future release then this is a non-issue in my book.  I mainly wanted to point out the behavior of swmod_o to make sure that it intended to behave that way.

Thank you,

Ryan

sdnellen, 2017-09-27:
Yes, was considering registering swmod output (but decided against it ;).  If had to start over again, delaying swmod by a cycle would likely have been a better baseline, but at this point changing the swmod/data relationship would be disruptive.  Just fyi, there is a pio module wrapper utility that has been beefed up a bit (w/ intent of handling simple async crossings, etc) and could be used in this case.  Eg, adding

      output systemverilog {
             wrapper_info {
                  add_sync_stages "*swmod_o" 1 // add sync delays to these outputs (use default clock)
             }
      }
to the parameters should add a 1 cycle delay to all swmod outputs in the wrapper module.  Feature is new and a bit of a work in progress tho.

closed 2017-09-27 by sdnellen