Juniper/open-register-design-tool #29: Why -verilog argument is missing

zhangpengsjtu, 2017-10-06:
Is ORDT able to generate verilog output anymore?

sdnellen, 2017-10-06:
It should still be there.  Is using -verilog not working?  If so, what version are you running.  The -verilog option IS missing in the display of parameter options, so will need to update this (it was always a hidden option).

zhangpengsjtu, 2017-10-08:
Thanks. I am using the latest one which I just cloned yesterday. I tried with -verilog, and it works.

May I know why it is always a hidden option? not recommended?

sdnellen, 2017-10-08:
I was hidden because we originally only used the systemverilog output and
just forgot to uncomment the option.  It's ok to use - really the code is
99% shared with the systemverilog output.  Plus, I often use -verilog for
debug using icarus, so option isn't going anywhere.  I'll make -verilog
visible in next release.

closed 2017-10-09 by zhangpengsjtu