Juniper/open-register-design-tool #30: enumerate information is missing in generated XML

zhangpengsjtu, 2017-10-23:
I have below enum and field defined in my systemRDL file, but after I run ORDT, I found that there is no enum information in the output xml file. 

May I know any option to get the enum dumped in output XML file, or it is not supported?

For example 
        enum ERRSLICE0_MISSIONERR_ENABLE_ERR25_enum{
           DISABLE = 1'd0;
           ENABLE = 1'd1;
        };
        field {encode=ERRSLICE0_MISSIONERR_ENABLE_ERR25_enum;sw=rw;hw=r;reset=1'h1;} ERR25[25:25];

sdnellen, 2017-10-23:
updated to add enum info to xml output in v171023.02.  will be picked up in next release.

zhangpengsjtu, 2017-10-24:
May I know when will be the next release, as we are going to use ORDT in one of our active open source projects

sdnellen, 2017-10-25:
Probably next week - you can always create a jar from latest source though.

zhangpengsjtu, 2017-10-25:
Thanks, I will watch ordt, and wait for the next release

closed 2017-10-25 by zhangpengsjtu