Juniper/open-register-design-tool #31: Enum is still not correct in XML file in release of 171102.01

zhangpengsjtu, 2017-11-03:
Hi, 

I have below enum for a field in systemRDL:

        enum S_STATUS_STATUS_0_enum {
            IDLE = 2'd0;
            RUNNING = 2'd1;
            PENDING = 2'd2;
        };

But in the generated XML file, I see the elem_name is encode_0x0/1/2, while it should be IDLE/RUNNING/PENDING instead. 
Did I miss some configuration in parms file?

        <enum_encode>
          <enc_name>S_STATUS_STATUS_0_enum</enc_name>
          <enc_elem>
            <enc_elem_name>encode_0x0</enc_elem_name>
            <enc_elem_value>0x0</enc_elem_value>
          </enc_elem>
          <enc_elem>
            <enc_elem_name>encode_0x1</enc_elem_name>
            <enc_elem_value>0x1</enc_elem_value>
          </enc_elem>
          <enc_elem>
            <enc_elem_name>encode_0x2</enc_elem_name>
            <enc_elem_value>0x2</enc_elem_value>
          </enc_elem>
        </enum_encode>

sdnellen, 2017-11-03:
Bug in xml gen code - try 171103.01

zhangpengsjtu, 2017-11-03:
Thanks, I will have a try

zhangpengsjtu, 2017-11-04:
Thansk, I see the correct ENUM in new release.

close issue

closed 2017-11-04 by zhangpengsjtu