Juniper/open-register-design-tool #36: Add support for callbacks to cppmod output

ebertland, 2017-11-09:
We need to register callback functions that are triggered on reads and/or writes of specific registers. The most common use cases are:

1) After a write is performed, copy or transform data written into one or more fields from the register model to another part of the reference model.
2) After a write is performed, trigger an action elsewhere in the reference model such as a FIFO push.
3) When a read is performed, modify the state of one or more fields in the register model before returning data to the caller.
4) After a read is performed, modify the state of a register such as clearing interrupt status bits.
5) After a read is performed, trigger an action elsewhere in the reference model such as a FIFO pop.

I propose 4 callback hooks: pre_write, post_write, pre_read, post_read.

post_write: use case 1) & 2)
pre_read: use case 3)
post_read: use case 4) & 5)

I cannot think of a use case for pre_write, but I think it should be added for consistency.

I plan to extend ordt_addr_elem so that callbacks may be registered with any ordt_reg or ordt_regset derived class. I don't have any need to register a callback on a field.

I'll submit a PR when the first draft is done.

ebertland, 2017-11-11:
My work in progress is in fork Cerebras/open-register-design-tool, branch jeff/cppmod-callbacks. I will use the new callback interfaces for a while to see if anything needs to be added or changed. Feedback is welcome.

sdnellen, 2017-11-29:
Good feature - lack of field level callbacks seems fine.  Submit PR when ready.

sdnellen, 2019-07-25:
Kamino cloned this issue to [sdnellen/open-register-design-tool](https://github.com/sdnellen/open-register-design-tool/issues/2)