Juniper/open-register-design-tool #37: Files in repo use a mix of line endings (CRLF vs LF)

ebertland, 2017-11-11:
I am developing on Linux. According to file, some files have CRLF and some do not (which means they only have LF). My editor automatically detects and switches modes, so it's not a big problem usually. The problem occurs when git thinks a file has been converted from one style to the other. Then, you get a huge diff in which every line in the file has been removed and added back. For an example, look at Ordt.java in commit ad8e9e69f2abb7b3c99102a0b727a05bf8db9f9c.

I have core.autocrlf=input, and I think that is the right setting to use on Linux. Setting that to true masks the problem by forcing every file to CRLF. I don't think that is the right solution.

Here's a useful description of the problem and what to do about it: https://stackoverflow.com/questions/170961/whats-the-best-crlf-carriage-return-line-feed-handling-strategy-with-git

```
$ find . -name '*.java' |xargs file
./src/ordt/parameters/ExtStringParameter.java:                                ASCII text
./src/ordt/parameters/ExtBooleanParameter.java:                               ASCII text
./src/ordt/parameters/ExtIntegerParameter.java:                               ASCII text
./src/ordt/parameters/ExtParameters.java:                                     ASCII text
./src/ordt/parameters/ExtRegNumberParameter.java:                             ASCII text
./src/ordt/parameters/ExtStringListParameter.java:                            ASCII text
./src/ordt/parameters/Utils.java:                                             C source, ASCII text, with CRLF line terminators
./src/ordt/parameters/ExtStringMapParameter.java:                             ASCII text
./src/ordt/parameters/ExtParameter.java:                                      ASCII text
./src/ordt/extract/DebugController.java:                                      ASCII text
./src/ordt/extract/Ordt.java:                                                 C source, ASCII text
./src/ordt/extract/DefinedProperty.java:                                      ASCII text
./src/ordt/extract/RegNumber.java:                                            ASCII text, with CRLF line terminators
./src/ordt/extract/PropertyList.java:                                         ASCII text, with CRLF line terminators
./src/ordt/extract/RdlModelExtractor.java:                                    ASCII text
./src/ordt/extract/model/ModBaseComponent.java:                               ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModEnum.java:                                        ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModEnumElement.java:                                 ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModSignal.java:                                      ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModComponent.java:                                   ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModAddressableInstance.java:                         ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModRegister.java:                                    ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModInstance.java:                                    ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModField.java:                                       ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModRootComponent.java:                               ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModRegSet.java:                                      ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModFieldSet.java:                                    ASCII text, with CRLF line terminators
./src/ordt/extract/model/ModIndexedInstance.java:                             ASCII text, with CRLF line terminators
./src/ordt/extract/DefinedProperties.java:                                    ASCII text
./src/ordt/extract/RegModelIntf.java:                                         ASCII text, with CRLF line terminators
./src/ordt/extract/JSpecModelExtractor.java:                                  ASCII text
./src/ordt/extract/OverlayFileInfo.java:                                      ASCII text
./src/ordt/output/RhsExpression.java:                                         C source, ASCII text
./src/ordt/output/RegProperties.java:                                         ASCII text, with very long lines, with CRLF line terminators
./src/ordt/output/InstanceProperties.java:                                    ASCII text, with CRLF line terminators
./src/ordt/output/RhsReference.java:                                          ASCII text, with CRLF line terminators
./src/ordt/output/verilog/VerilogTestBuilder.java:                            ASCII text, with CRLF line terminators
./src/ordt/output/verilog/VerilogBuilder.java:                                ASCII text, with CRLF line terminators
./src/ordt/output/othertypes/XmlBuilder.java:                                 ASCII text
./src/ordt/output/othertypes/RegListBuilder.java:                             ASCII text, with CRLF line terminators
./src/ordt/output/othertypes/JsonBuilder.java:                                ASCII text
./src/ordt/output/othertypes/JspecBuilder.java:                               ASCII text
./src/ordt/output/othertypes/RdlBuilder.java:                                 ASCII text, with CRLF line terminators
./src/ordt/output/RegSetProperties.java:                                      ASCII text, with CRLF line terminators
./src/ordt/output/FieldSetProperties.java:                                    ASCII text, with CRLF line terminators
./src/ordt/output/systemverilog/SystemVerilogChildInfoBuilder.java:           ASCII text
./src/ordt/output/systemverilog/io/SystemVerilogIOStruct.java:                ASCII text
./src/ordt/output/systemverilog/io/SystemVerilogIOSignalList.java:            ASCII text
./src/ordt/output/systemverilog/io/SystemVerilogIOElement.java:               ASCII text
./src/ordt/output/systemverilog/io/SystemVerilogIOInterface.java:             ASCII text
./src/ordt/output/systemverilog/io/SystemVerilogIOSignalSet.java:             ASCII text, with very long lines
./src/ordt/output/systemverilog/io/SystemVerilogIOSignal.java:                ASCII text
./src/ordt/output/systemverilog/SystemVerilogLogicModule.java:                ASCII text
./src/ordt/output/systemverilog/SystemVerilogDefinedSignal.java:              ASCII text
./src/ordt/output/systemverilog/SystemVerilogBuilder.java:                    ASCII text, with CRLF line terminators
./src/ordt/output/systemverilog/SystemVerilogDefinedSignals.java:             ASCII text
./src/ordt/output/systemverilog/SystemVerilogTestBuilder.java:                ASCII text, with CRLF line terminators
./src/ordt/output/systemverilog/common/SystemVerilogSignal.java:              ASCII text, with CRLF line terminators
./src/ordt/output/systemverilog/common/SystemVerilogSignalList.java:          ASCII text, with CRLF line terminators
./src/ordt/output/systemverilog/common/SystemVerilogCoverGroups.java:         ASCII text, with CRLF line terminators
./src/ordt/output/systemverilog/common/SystemVerilogModule.java:              ASCII text
./src/ordt/output/systemverilog/common/SystemVerilogInstance.java:            ASCII text
[... 13 lines cut ...]
./src/ordt/output/drvmod/DrvModRegSetInstance.java:                           ASCII text, with very long lines
./src/ordt/output/drvmod/DrvModBuilder.java:                                  ASCII text, with very long lines, with CRLF line terminators
./src/ordt/output/SignalProperties.java:                                      ASCII text, with CRLF line terminators
./src/ordt/output/uvmregs/UVMRegsBuilder.java:                                ASCII text, with CRLF line terminators
./src/ordt/output/uvmregs/UVMRdlLite1Classes.java:                            ASCII text
./src/ordt/output/uvmregs/UVMRegsCommon.java:                                 ASCII text
./src/ordt/output/uvmregs/UVMRegsTranslate1Builder.java:                      ASCII text
./src/ordt/output/uvmregs/UVMRegsLite1Builder.java:                           ASCII text
./src/ordt/output/uvmregs/UVMRdlClasses.java:                                 ASCII text, with CRLF line terminators
./src/ordt/output/uvmregs/UVMRdlTranslate1Classes.java:                       ASCII text
[... cut ...]
./src/ordt/output/FieldProperties.java:                                       ASCII text, with CRLF line terminators
./src/ordt/output/JspecCategory.java:                                         C source, ASCII text
./src/ordt/annotate/AnnotateShowCommand.java:                                 ASCII text
./src/ordt/annotate/AnnotateCommand.java:                                     ASCII text
./src/ordt/annotate/AnnotateSetCommand.java:                                  ASCII text
./src/ordt/annotate/AnnotateNullCommand.java:                                 ASCII text
```

sdnellen, 2017-11-12:
Think the core.autocrlf=input is correct setting for linux/osx... it should change endings to lf on commit and then pull files as-is (see autocrlf secion at https://git-scm.com/book/en/v2/Customizing-Git-Git-Configuration).  Issue is that the ordt github repo is a mess wrt line endings - at some point I moved dev from windows to osx and didnt see issue as editor cleaned these up transparently.  Now that I've switched to autocrlf=input, files are starting to be stored with lf only, which is desired behavior.  Can either allow these to clean up over time or explicitly push all files to clean this up.  Also fyi, am using egit/eclipse which apparently has an issue with .gitattributes.  Think we'll just need to be diligent in insuring that contributors have correct autocrlf setting.

ebertland, 2017-11-14:
I usually prefer one big commit to clean-up issues like this (line endings, indentation, style issues, etc.). Now that I know what's happening, I don't have a problem with it, so it's your call.

closed 2018-02-05 by sdnellen