Juniper/open-register-design-tool #46: Binding RDL UDP to addrmap is not recognized

amykyta3, 2018-04-28:
If I do the following:
```
property my_rdl_udp {
  type=boolean;
  component=addrmap;
};

addrmap my_sub_addrmap {
    my_rdl_udp;
    reg {} a;
};

addrmap my_top_addrmap {
    my_sub_addrmap sub;
} top;
```

It seems that the RDL user-defined property is discarded, and does not show up in the XML output.
Instead the following warning is emitted:
```
*** WARNING ***: assignment of property my_rdl_udp will be ignored in regfile my_sub_addrmap near line 10
```

If I switch the UDP declaration to be bound to a `regfile` component, it works correctly, even though the target component is actually an addrmap.
So far to future proof, my workaround has been to bind to both component types:
```
component=regfile | addrmap;
```

sdnellen, 2018-04-30:
fix committed in 180430.01. will be picked up in next rls.

amykyta3, 2018-05-07:
Confirmed fix in pre-release 180502.01

closed 2018-05-07 by sdnellen