Juniper/open-register-design-tool #48: param min_data_size seems to be interpreted as the default register size for SystemRDL

amykyta3, 2018-05-07:
Interpretation of the global parameter "min_data_size" seems to have changed in the latest release.

If left undefined, SystemRDL 1.0 defines the default register as 32-bits.
The min_data_size ORDT parameter seems to override this and coerce default register widths to that value.

Testcase below:
```
<PARMS>
global {
  min_data_size = 8
}
</PARMS>

addrmap {
    reg {
        field {
            sw=r; hw=w;
        } F1[7:0];
        field {
            sw=r; hw=w;
        } F2[15:8];
    } MY_REG @ 0x0;
} MY_ADDRMAP;
```

Result in:
```
*** ERROR ***: Unable to fit all fields in 8b register instance MY_REG
```

sdnellen, 2018-05-16:
Fix added in 180514.01.  Will be picked up in next release.

sdnellen, 2018-10-17:
Fix picked up in 181009.01 rls.

closed 2018-10-17 by sdnellen