Juniper/open-register-design-tool #49: Using native uvm classes without extended classes

Bobbyq66, 2018-05-08:
Hello,

Allow an option to generate uvm registers with native UVM class register, uvm_reg and uvm_reg_fields, as part of uvm_reg_block;

The capabilities of lite1 is limited but heavy1 is too much such that you would need to change existing UVM RAL environments because it is already using uvm_reg_block;

This way ordt can be used to generate uvm reg blocks for existing old environments and also be able to accommodate new uvm reg flow with extended rdl classes.

Thanks, Bo

ebertland, 2018-05-08:
What's wrong with using the extended _rdl classes? They can be used with existing environments because they are derived from uvm_reg* base classes.

Bobbyq66, 2018-05-08:
Thanks for the quick reply.

There are issues at run time because as an example I completely change one of our uvm_blocks to extended rdl classes.

Just a couple of examples, I haven't debugged too much but I was hoping for a plug and play integration into our ral sequences. 

In this example, apb_timer_reg_block was changed to using _rdl classes. However, apb_subsys_reg_block is still uvm_reg_block; As intended, we dont want to re-generate all existing working uvm_reg_blocks to support _rdl.

Thus you cannot cast back two different types;

```
          $cast(rdl_parent, get_parent());
              |
ncsim: *E,BCLCST (/project/mpp/bqin/mpp/sources/dv/uvc/mpp_ral/pkg.sv,78|14): Invalid cast: a value with the class datatype 'apb_subsys_ral_pkg::apb_subsys_reg_block' cannot be assigned to a class variable with the datatype 'ordt_uvm_reg_pkg::uvm_reg_block_rdl'.
ncsim: *E,TRNULLID: NULL pointer dereference.
          File: /project/mpp/bqin/mpp/sources/dv/uvc/mpp_ral/pkg.sv, line = 70, pos = 28
         Scope: worklib.ordt_uvm_reg_pkg::uvm_reg_block_rdl::get_rdl_name
          Time: 0 FS + 9
```
In this example, we use reg_map to append block submaps. Not sure why this class is not existing.

```
reg_map.add_submap(apb_timer_reg_block[i].reg_map, 32'h00000000 + (i*32'h1000));
                                                      |
ncvlog: *E,NOTCLM (/project/mpp/bqin/mpp/sources/dv/uvc/mpp_ral/apb_subsys_ral/apb_subsys_reg_block.svh,75|54): reg_map is not a class item.
	package worklib.apb_subsys_ral_pkg:sv

```

Thanks, Bo

sdnellen, 2018-06-19:
Checked in a first cut in 180619.01 (uvm_model_mode = native).

Bobbyq66, 2018-06-21:
Thanks Scott, verified to be a simple integration into our current uvm_regs. Closing.

closed 2018-06-21 by Bobbyq66