Juniper/open-register-design-tool #50: Feature Request : There is no way to generate the c++ driver with more than two RDL files

kprabhu36, 2018-08-22:
1. There is no way to generate c++ driver files (using -overlay) with more than two RDLs.
2. It will be nice to have an option to generate a single c++ driver with multiple RDL as a input. This can be useful for a block which has more sub-blocks/RDLs.

sdnellen, 2018-08-23:
A fix to allow multiple **-overlay** tags on the command line has been committed in 180823.01 and will be picked up in the next release.

Note that intent the overlay option was to handle the case of multiple versions of a register space in an overlapping address region, not really for non-overlapping sub-blocks within a larger space - it may be fine for your application, though.  For the case of generating a single model containing multiple subblocks, would recommend using a preprocessor tool (https://github.com/Juniper/open-register-design-tool/wiki/Running-Ordt#input-file-pre-processors) to include sub-module rdl, define offsets, and produce a sigle output rdl file that can be used to create the model (without needing overlays).  Something like:

full_block.pre_rdl

      #include "submod1.rdl"
      #include "submod2.rdl"
      #include "submod3.rdl"
      addrmap {
            submod1 submod1 @0x0;
            submod2 submod2 @0x1000;
            submod2 submod3 @0x2000;
      } full_block;

kprabhu36, 2018-08-23:
Thank you. Will test it out.

sdnellen, 2018-10-17:
Fixed in 181009.01 rls.

closed 2018-10-17 by sdnellen