Juniper/open-register-design-tool #54: Non-synthesizable "#1" in generated rtl

kdloe, 2019-03-05:
Is there a option to not have the "#1" in the generated rtl?

  //------- reg assigns for nsg_axirdl_mdb_shm_intm
  always_ff @ (posedge clk or negedge sig_ordtreg_rst_n) begin
    if (! sig_ordtreg_rst_n) begin
      rg_nsg_axirdl_mdb_shm_intm_err_int <= #1 1'd0;
    end
    else begin
      rg_nsg_axirdl_mdb_shm_intm_err_int <= #1  reg_nsg_axirdl_mdb_shm_intm_err_int_next;
    end
  end

sdnellen, 2019-03-05:
No option to remove these currently.  What synth tool has issues with #1?

kdloe, 2019-03-12:
Not that a synth tool has an issue with it, it shows up in our LINT checks as non-synthesizable and may contribute to sim vs silicon mismatches. We have a mix of legacy verilog code and newer system verilog code and do not allow #1 due to a past history of simulators that had issues with delta delays. Even though today's tools have less issues, our methodology is that the UVCs take care of any timing issues at the verif boundary. We also do not want surprise timescale differences.

sdnellen, 2019-03-13:
Added include_sequential_assign_delays param in v190312 - set to false to remove these pound 1 delays (seems to gen correctly but not tested).

sdnellen, 2019-05-28:
Try release 190524.01.

neenuprince, 2019-06-06:
Tried the new Release and we don't see the #1's anymore

closed 2019-06-07 by kdloe