Juniper/open-register-design-tool #56: Support for "sharedextbus" and "errextbus"

kdloe, 2019-03-05:
The ORDT wiki documentation lists "sharedextbus" and "errextbus" as unsupported attributes. When can/will these become supported attributes? :-)

sdnellen, 2019-03-05:
Not this week ;)   ...or anytime soon realistically. 

Currently tool only supports the following aggregations for creating external regions:
- single regs/reg arrays w/ external tag
- single regfiles/regfile arrays w/ external tag
- child addrmaps (treated as an external regfile)
- single regs/reg arrays within a replicated regfile hierarchy w/ external tag and rep_level option

It isn't clear what the errextbus actually does from the rdl1.0 spec, so it wasn't implemented.  However, all generated external interfaces have a nack input by default for indicating an invalid transaction occurred - this indication is returned to the main decoder interface.

kdloe, 2019-03-12:
Hello - We use the shared external bus to connect to multiple SRAMS that have been mapped into the register space and save on signals. We also use the errextbus for returning the SRAM secded errors.

sdnellen, 2019-03-13:
Currently you will need to wrap elements in a single regfile to get a single external interface.  Only exception is case of regs in replicated regfile hierarchy, which can be collapsed to single i/f using rep_level external option.

Can return nack from external region for ecc/parity fails / ack for valid transactions.  Generally use interrupt scheme for fail detect/logging of ram fails.

kdloe, 2019-03-26:
Tried this out and we think it will work for us.

closed 2019-03-26 by kdloe