Juniper/open-register-design-tool #57: Internal error accessing SystemVerilogDefinedSignalMap generating UVM output

ebertland, 2019-04-11:
This error occurs when generating UVM register model output only.

```
$ ordt -uvmregs out/test_uvm_reg.sv test.rdl
Open Register Design Tool, version=190321.01, input=test.rdl
*** WARNING ***: No parameters file specified.  Default or inline defined parameters will be used.
Ordt: building UVM regs...
*** ERROR ***: Attempted to access SystemVerilogDefinedSignalMap prior to initialization - insure map is defined in concrete child class
Ordt exited due to error Wed Apr 10 18:08:00 PDT 2019
```

A work-around is to also generate RTL using `-systemverilog`.

```$ ordt -systemverilog out/test_rtl/ -uvmregs out/test_uvm_reg.sv test.rdl
Open Register Design Tool, version=190321.01, input=test.rdl
*** WARNING ***: No parameters file specified.  Default or inline defined parameters will be used.
Ordt: building systemverilog...
Ordt: writing systemverilog file out/test_rtl/my_cntrs_pio.sv...
Ordt: writing systemverilog file out/test_rtl/my_cntrs_jrdl_logic.sv...
Ordt: writing systemverilog file out/test_rtl/my_cntrs_jrdl_decode.sv...
Ordt: building UVM regs...
Ordt: writing UVM regs file out/test_uvm_reg.sv...
Ordt complete Wed Apr 10 18:05:53 PDT 2019
```

[test.zip](https://github.com/Juniper/open-register-design-tool/files/3066483/test.zip)

ebertland, 2019-04-11:
Commenting out `u.value->incr = l.value->overflow;` avoids the problem, so it is related to signal assignments. That's probably obvious based on the error message.

ebertland, 2019-04-11:
This test fails on HEAD of master (550847c19724fb9654a8a01dde6d9b77d8b81d6f) and on the latest release jar (version 181009.01).

ebertland, 2019-04-11:
I confirmed with git bisect that this bug was introduced in 669b0217b30c4513018b926aacf582af5e937766.

sdnellen, 2019-04-12:
Committed fix in 190411.01.

ebertland, 2019-04-28:
The new version works for me. Closing.

closed 2019-04-28 by ebertland