Juniper/open-register-design-tool #58: Code generated by "resetsignal" attribute not proper

neenuprince, 2019-04-17:
Hi, 
We noticed that the code generated when the resetsignal attribute is set for a single field in a register is not proper.
In the below code 
the registerfield "fieldrstextsig has the "resetsignal = sig_test_rst_sig" attribute set.
and the system reset is sig_ordtreg_rst_n.
_always_ff @ (posedge clk or posedge sig_test_rst_sig or negedge sig_ordtreg_rst_n) begin
if (sig_test_rst_sig) begin
      rg_other_rf_attr_packedfields_fieldrstextsig <= #1 2'b0;
    end
    else if (! sig_ordtreg_rst_n) begin
      rg_other_rf_attr_packedfields_swrclr <= #1 2'h0;
     end
    else begin
      rg_other_rf_attr_packedfields_swrclr <= #1  reg_other_rf_attr_packedfields_swrclr_next;     
  end_
in the above code on a system reset, if the sig_test_rst_sig is asserted, the "rg_other_rf_attr_packedfields_swrclr" field does not get reset.

Is there a guideline that, we cannot have resetsignal attribute to a single field of a register?

sdnellen, 2019-04-18:
Looks like a bug.  Can you fwd simple rdl (including reset signal define, resetsignal assign, etc) and your parameter settings for this case and note the version you are running?

sdnellen, 2019-04-21:
Able to recreate issue - fix committed in 190421.01 and will be picked up in next release.

sdnellen, 2019-05-28:
Try release 190524.01.

closed 2019-05-28 by neenuprince