Juniper/open-register-design-tool #61: Registers with intr attribute and bothedge and negedge not reseting properly

kdloe, 2019-05-20:
We are testing two of the interrupt register types with (bothedge intr) and (negedge intr) and we are seeing that the X's at reset are not cleared out. The posedge and level are functioning properly.

**Sample RDL:**

    reg negintr_t {
        name = "Reg negedge intr";
        desc = "A register with negedge intr";
             default hw = na; // Ordt have as na
            default sw = rw;
        field {
            name = "negedge_intr check";
            desc = "checking that the negedge intr works";
            negedge intr;
            woclr;
        } interrupt = 1'b0;
    };

    reg bothintr_t {
        name = "Reg bothedge intr";
        desc = "A register with bothedge intr";
             default hw = na; // Ordt have as na
            default sw = rw;
        field {
            name = "bothedge_intr check";
            desc = "checking that the bothedge intr works";
            bothedge intr;
            woclr;
        } interrupt = 1'b0;
    };

    attr_negintr_t           negintr;
    attr_bothintr_t          bothintr;

**Sample generated Code** 

  //------- combinatorial assigns for bothintr
  always_comb begin
    reg_bothintr_interrupt_next = rg_bothintr_interrupt;
    l2h_bothintr_intr_o = 1'b0;
    intr_bothintr_clear = 1'b0;
    reg_bothintr_interrupt_next = ((h2l_bothintr_interrupt_intr ^ intr_bothintr_interrupt_previntr) | rg_bothintr_interrupt);
    l2h_bothintr_intr_o = l2h_bothintr_intr_o | (rg_bothintr_interrupt);
    intr_bothintr_clear = intr_bothintr_clear | (intr_bothintr_interrupt_delay & ~rg_bothintr_interrupt);
    l2h_bothintr_intr_o = l2h_bothintr_intr_o & ~intr_bothintr_clear;
    if (d2l_bothintr_we) reg_bothintr_interrupt_next = (reg_bothintr_interrupt_next & ~d2l_bothintr_w [0] );
  end
  
  //------- reg assigns for bothintr
  always_ff @ (posedge clk or negedge sig_ordtreg_rst_n) begin
    if (! sig_ordtreg_rst_n) begin
      rg_bothintr_interrupt <= #1 1'b0;
    end
    else begin
      rg_bothintr_interrupt <= #1  reg_bothintr_interrupt_next;
      intr_bothintr_interrupt_previntr <= #1 h2l_bothintr_interrupt_intr;
      intr_bothintr_interrupt_delay <= #1 rg_bothintr_interrupt;
    end
  end

  //------- combinatorial assigns for negintr
  always_comb begin
    reg_negintr_interrupt_next = rg_negintr_interrupt;
    l2h_negintr_intr_o = 1'b0;
    intr_negintr_clear = 1'b0;
    reg_negintr_interrupt_next = ((intr_negintr_interrupt_previntr & ~h2l_negintr_interrupt_intr) | rg_negintr_interrupt);
    l2h_negintr_intr_o = l2h_negintr_intr_o | (rg_negintr_interrupt);
    intr_negintr_clear = intr_negintr_clear | (intr_negintr_interrupt_delay & ~rg_negintr_interrupt);
    l2h_negintr_intr_o = l2h_negintr_intr_o & ~intr_negintr_clear;
    if (d2l_negintr_we) reg_negintr_interrupt_next = (reg_negintr_interrupt_next & ~d2l_negintr_w [0] );
  end
  
  //------- reg assigns for negintr
  always_ff @ (posedge clk or negedge sig_ordtreg_rst_n) begin
    if (! sig_ordtreg_rst_n) begin
      rg_negintr_interrupt <= #1 1'b0;
    end
    else begin
      rg_negintr_interrupt <= #1  reg_negintr_interrupt_next;
      intr_negintr_interrupt_previntr <= #1 h2l_negintr_interrupt_intr;
      intr_negintr_interrupt_delay <= #1 rg_negintr_interrupt;
    end
  end

sdnellen, 2019-05-21:
Please provide additional info for recreating this issue.  Added your code to the rdl_intr_01 test and didn't see any problems with the edge intrs (see test/basic_tests/rdl_intr_01).  If you generate rtl and testbench for this test using rdl/parms, you can simulate the  simple test.

sdnellen, 2019-05-28:
Try release 190524.01.

sdnellen, 2019-07-25:
Kamino cloned this issue to [sdnellen/open-register-design-tool](https://github.com/sdnellen/open-register-design-tool/issues/4)

kdloe, 2019-08-16:
Not able to reproduce this with the latest release.

closed 2019-08-16 by kdloe