Juniper/open-register-design-tool #65: Unused wires generated

sjalloq, 2019-06-11:
Hi, I'm testing out an interrupt register that is set by hardware and cleared by a software read.  I noted that along with the _intr input from the HW, there is also a redundant write path from the software side generated.

In the following Verilog output, the inputs d2l_u_1_my_irq_0_w* to the root address map are unused.

[regfile.v.txt](https://github.com/Juniper/open-register-design-tool/files/3277249/regfile.v.txt)

[regfile.rdl.pp.final.txt](https://github.com/Juniper/open-register-design-tool/files/3277245/regfile.rdl.pp.final.txt)

sdnellen, 2019-06-11:
ordt uses a general interface between decoder and logic submodules for all registers - it is assumed that synthesis will clean these up.  It is key to not retain submodule hierarchy in synthesis as in addition to unused signals, there are a number of replicated common signals that will have non-optimum structure if sub-level ports are retained.  Need to add a note to the docs re this behavior.

sjalloq, 2019-06-12:
OK, thanks.

closed 2019-06-12 by sjalloq