Juniper/open-register-design-tool #66: Reset terms are not the correct width

sjalloq, 2019-06-13:
I found one example of a register reset term not being the correct width.  This might be a case of me using a generic single bit field definition and expanding that in the reg fefinition.

In the following RDL, the register _vendor is the correct with of 8-bits but its reset term is 1'b0.

```
field interrupt_enable {
      hw=na ;         // Hardware has no access
      sw=rw ;         // Software can read/write
      reset = 1'b0 ;  // Reset value is zero
};

reg irq_en_reg_8 { 
    regwidth = 16 ;

    name = "SFF-8636 Interrupt Enable Flags Byte 8" ;
    desc = "These interrupts are vendor specific" ;

    interrupt_enable _vendor[8] ;
};
```

sdnellen, 2019-06-13:
Yes, tool currently doesn't resize the width of reset constant once the width is set in the field def assign.  For this case could try using int assign, ie *reset = 0;*  (have not tested)

sjalloq, 2019-06-14:
Thanks.  That didn't work so I'll stick it inside a for loop.

closed 2019-06-14 by sjalloq