Juniper/open-register-design-tool #67: Interrupt enable definitions are not obeyed

sjalloq, 2019-06-13:
I've followed the example in the RDL spec on how to generate hierarchical interrupts.  At the top level I have two registers with a single bit in each, one a global interrupt and one a global enable.  I've tried using both the 'intr' and 'nonsticky intr' properties on the interrupt and it behaves the same way.  How am I supposed to mask this top level interrupt?

```
reg ctrl_reg {
    regwidth = 16 ;
    name = "Global Control Register" ;
    
    default hw=r ;
    default sw=rw ;
    default reset = 1'b0 ;
    
    field {
      desc = "Global enable for all interrupts" ;
      hw=na;
    } global_irq_en ;
};

reg status_reg {
    regwidth = 16 ;
    name = "Global Status Register" ;

    default hw=w ;
    default sw=r ;
    default reset = 1'b0 ;

    field {
      desc = "Global interrupt flag" ;
      hw=na ;
      intr;
    } global_irq ;
};

addrmap address_map {
	// Instantiate the generic control and status regs
	ctrl_reg    _control  @0x000 ;
	status_reg  _status   @0x002 ;

	// Mask the global interrupt enable
	_status.global_irq->enable = _control.global_irq_en ;
};
```

That results in:
```
  //------- combinatorial assigns for _status
  always @ (*) begin
    reg__status_global_irq_next = rg__status_global_irq;
    l2h__status_intr_o = 1'b0;
    l2h__status_intr_o = l2h__status_intr_o | (rg__status_global_irq);
    reg__status_global_irq_next = l2h__sff_irq_status_intr_o;
  end
  
  //------- reg assigns for _status
  always @ (posedge clk) begin
    if (! sig_reset_n) begin
      rg__status_global_irq <= #1 1'b0;
    end
    else begin
      rg__status_global_irq <= #1  reg__status_global_irq_next;
    end
  end
 ```

sjalloq, 2019-06-13:
And is that another typo in your output generator?  Shouldn't the register name and the _next term be the same?  Looks like you missed the 'e' from 'reg' in rg__status_global_irq.

sdnellen, 2019-06-13:
Not able to recreate this issue - directly copied your code (older version i think that didnt have the ->next assign) and see the following for _status assigns:

        //------- combinatorial assigns for _status
          always @ (*) begin
            reg__status_global_irq_next = rg__status_global_irq;
            l2h__status_intr_o = 1'b0;
            reg__status_global_irq_next = (h2l__status_global_irq_intr | rg__status_global_irq);
            l2h__status_intr_o = l2h__status_intr_o | (rg__status_global_irq & rg__control_global_irq_en);
          end

Assume you see no errors/warnings when running?  If not, might try removing the comment prior to the ->enable assign (just a guess)

Register name starting with "rg_" is no typo - just a way to allow easy identification of actual field registers by name.

sjalloq, 2019-06-14:
OK, I've attached a testcase with my full RDL.  Untar it and run 'make'.  The archive has the jar file and the preprocessor.

Thanks.
[testcase.tar.gz](https://github.com/Juniper/open-register-design-tool/files/3289889/testcase.tar.gz)

sjalloq, 2019-06-17:
Thanks a lot.  If you could fire off a build that would be useful as I can't currently build from source.  See #63

sjalloq, 2019-06-18:
Confirmed fixed.  Thanks a lot.

closed 2019-06-18 by sjalloq