Juniper/open-register-design-tool #69: Register field ordering is broken.

sjalloq, 2019-07-02:
The RDL spec shows how fields are packed in a register on p.44 of the v1.0 spec.  In the case of LSB ordering, the following should hold:

```
lsb0;
reg {
field {} A; // Single bit from 0 to 0
field {} B[3]; // 3 bits from 3:1
// 4 bits from 7 to 4 are reserved and unused
field {} C[15:8]; // 8 Bits from 15 to 8
field {} C[5]; // 5 Bits from 20 to 16
};
```

However, this doesn't appear to be the case with Ordt.  My understanding from the above example is that the following register:

```
reg irq_reg_6 { 
    regwidth = 16 ;
    interrupt _init_done ;
    interrupt _tc_ready ;
    interrupt _temp_low_warn[4:4] ;
    interrupt _temp_high_warn ;
    interrupt _temp_low_alarm ;
    interrupt _temp_high_alarm ;
};
```

should result in a packed register of the form:

Bit 0 : init_done
Bit 1: tc_ready
Bit 2 : reserved
Bit 3 : reserved
Bit 4 : temp_low_warn
Bit 5 : temp_high_warn
Bit 6 : temp_low_alarm
Bit 7 : temp_high_alarm

Ordt is outputting the following:

Bit 0 : init done
Bit 1: tc_ready
Bit 2 : temp_high_warn
Bit 3 : temp_low_alarm
Bit 4 : temp_low_warn
Bit 5 : temp_high_alarm

sdnellen, 2019-07-02:
Yes, ordt will pack fields with no assigned offset in first available space in the reg starting from lsb.  Spec was somewhat ambiguous here so implementation may differ from other compilers - given volume of code out there, default behavior will not be changed.  Might be worth adding a switch at some point, but solution for now is to use bit offsets when instancing fields.

sdnellen, 2019-07-25:
Kamino cloned this issue to [sdnellen/open-register-design-tool](https://github.com/sdnellen/open-register-design-tool/issues/8)