Juniper/open-register-design-tool #70: RTl code has floating signals  for HWRead  SWwrite registers causing X's

neenuprince, 2019-07-18:
The .jrdl_logic.sv file generates a signal for the HW Read SW write register, that is floating.
the signal l2d_<reg>_r is the signal that remains floating, causing X's to propogate.
what I found is that this register genenrated code misses an assignment as below  
always_comb begin
    l2d_other_rf_attr_hwrsww_r = rg_other_rf_attr_hwrsww_hwrsww;
  end

Simple rdl is as below
//Register with hw = r and sw = w
    reg attr_hwrsww_t {
        name = "Hw r and sw w";
        desc = "register with hw = r and sw = w";
            default hw = r;
            default sw = w;
        field {
            fieldwidth = 32;
           // reset = 32'h0;
        } hwrsww = 32'h0;
    };

sdnellen, 2019-07-19:
See issue #65

sdnellen, 2019-07-25:
Kamino cloned this issue to [sdnellen/open-register-design-tool](https://github.com/sdnellen/open-register-design-tool/issues/9)