Juniper/open-register-design-tool #72: how to write addrmap block?

zhajio1988, 2019-08-17:
*** ERROR ***: No rdl structures instanced or typedefs specified for processing.
below is my addrmap block look likes:
`addrmap addrmap_PICO{
    name = "ATXMEGA SPI controller";
    desc = "Register description of Atmel XMEGA AU's SPI controller
    Transcribed from original manual as an example exercise:
    http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8331-8-and-16-bit-AVR-Microcontroller-XMEGA-AU_Manual.pdf";
    signal { activelow; field_reset; cpuif_reset;} reset_n;
    regs_ral_demo ral_regs_demo @0x0;
};
`

sdnellen, 2019-08-18:
ordt looks for a single addrmap instance at root level by default - if multiple components are instanced at root, only the first will be processed.  Format of an rdl addrmap component is...

      addrmap <optional_component_name> {
           …addrmap .contents
      } <optional_instance_name> ;

So, you could do either....

      addrmap  {
           …addrmap .contents
      } addrmap_PICO_instance_name  ;

Or....

      addrmap  addrmap_PICO {
           …addrmap .contents
      } ;
      addrmap_PICO addrmap_PICO_instance_name;

See the wiki for details/examples.

zhajio1988, 2019-08-19:
> ordt looks for a single addrmap instance at root level by default - if multiple components are instanced at root, only the first will be processed. Format of an rdl addrmap component is...
>
> ``` [...]

@sdnellen  so 
`
addrmap addrmap_PICO{ name = "ATXMEGA SPI controller"; desc = "Register description of Atmel XMEGA AU's SPI controller Transcribed from original manual as an example exercise: http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8331-8-and-16-bit-AVR-Microcontroller-XMEGA-AU_Manual.pdf"; signal { activelow; field_reset; cpuif_reset;} reset_n; regs_ral_demo ral_regs_demo @0x0; };
`
is wrong usage?

sdnellen, 2019-08-19:
The usage isn't incorrect, but it doesn't tell the tool what to run - keep in mind multiple addrmap components can be defined at root level and for some outputs, having no addrmap is a valid form.  The first root instance tells the tool what to run by default.  Alternately, an rdl input control parameter can be specified to process a specific component, eg *process_component = "addrmap_PICO"*.

zhajio1988, 2019-08-21:
@sdnellen 
thank you . i understand.

closed 2019-08-21 by sdnellen