Juniper/open-register-design-tool #74: "gated_logic_access_delay" affects access delay even if clock gating disabled

roowatt, 2019-08-29:
After some testing with the latest release I notice that there is a significant access delay, looking into the options I found the "gated_logic_access_delay" setting that impacted it, even when clock gating is disabled.

What is the expected access delay with the parallel option and clock gating disabled? 

What is the minimum I should be able achieve? How to configure for that?

sdnellen, 2019-08-31:
Unable to reproduce this issue.  With parallel interface and no **use_gated_logic_clock** or **gated_logic_access_delay** parameters specified, a delay of 2 cycles is observed between *pio_dec_read/write* and *dec_pio_ack* response when accessing an internal register in the root decoder (this is the min delay - see waves below).  If only the **gated_logic_access_delay** is specified, no change in behavior is observed.  If **use_gated_logic_clock** is specified as true, the *dec_pio_ack* will be delayed by an additional **gated_logic_access_delay** cycles. 

[image]

roowatt, 2019-09-02:
Thanks for looking into it, it looks like a problem in my build system.

closed 2019-09-02 by roowatt