Juniper/open-register-design-tool #76: gererate register verilog module has synthesis error when use Synopsys DC

zhajio1988, 2019-11-13:
when i use below configure file to generate verilog file
`global {
                                            // use_js_address_alignment = false
    allow_unordered_addresses = true
    suppress_alignment_warnings = true
}
input rdl {
    resolve_reg_category = true             // if register category is unspecified, try to determine from rdl
    restrict_defined_property_names = false
}

                                            // systemverilog output parameters
output systemverilog {
	leaf_address_size = 32                  // leaf address bits
	base_addr_is_parameter = false          // base address parameter will be added to top module
	use_gated_logic_clock = false           // use separate gated clock for registers
	always_generate_iwrap = false           // create int wrapper mod even if none specified in rdl
	suppress_no_reset_warnings = true       // inhibit field no reset messages
	include_default_coverage = false        // include default cover points in rtl
	use_async_resets = true                 // registers will use asynchronous reset
    root_decoder_interface = parallel       // parallel | parallel_pulsed | leaf | serial8 | ring8 | ring16 | ring32 
    optimize_parallel_externals = true
    include_sequential_assign_delays= false // not include #1 delay on sequential assigns    
}

                                            // jspec output parameters
output jspec {
	root_regset_is_instanced = true         // instance the root or make it a typedef?
}

                                            // reglist output parameters
output reglist {
	display_external_regs = true            // include external regs in output?
	show_reg_type = false                   // show int/ext type for each reg?
	show_fields = false                     // show field info for each reg
}

                                            // uvmregs output parameters
output uvmregs {
	suppress_no_category_warnings = true    // inhibit reg no category messages
	include_address_coverage = false        // include address coverage in model
	reuse_uvm_classes = false               // allow reuse of classes in model
    uvm_model_mode = native                 // set uvm model type (heavy | lite1 | native)
    regs_use_factory = true
}

                                            // xml output parameters
output xml {
    include_field_hw_info = true
}
`
will generate below code segment：
`  //------- reg assigns for pio i/f
  always @ (posedge clk or negedge sig_reset_n) begin
    if (! sig_reset_n) begin
      pio_write_active <=  1'b0;
      pio_read_active <=  1'b0;
    end
    else begin
      pio_write_active <=  pio_write_active ? pio_no_acks : pio_activate_write;
      pio_read_active <=  pio_read_active ? pio_no_acks : pio_activate_read;
    end
    pio_dec_address_d1 <=   pio_dec_address;
    pio_dec_write_data_d1 <=  pio_dec_write_data;
  end`
when use Synopsys DC systhesis this file will report below error：
Error:  design/registers/rtl/hwa_ldpc_enc_reg.v:1775: The statements in this 'always' block are outside the scope of the synthesis policy. Only an 'if' statement is allowed at the top level in this always block. (ELAB-302) 

@sdnellen  could you fix this issue？

sdnellen, 2019-11-14:
what version of ordt are you using?

zhajio1988, 2019-11-14:
190617.01
commit ac1633675f42c6758b428ec0e43e0685f9c6958c
i download the jar file from https://github.com/Juniper/open-register-design-tool/releases page.

zhajio1988, 2019-11-14:
> 190617.01
> commit [ac16336](https://github.com/Juniper/open-register-design-tool/commit/ac1633675f42c6758b428ec0e43e0685f9c6958c)
> i download the jar file from https://github.com/Juniper/open-register-design-tool/releases page.

@sdnellen

sdnellen, 2019-11-14:
will look into this issue - in the meantime, can you try downloading 170725.01 at  https://github.com/sdnellen/open-register-design-tool/releases to see if this bug persists.

zhajio1988, 2019-11-14:
hi @sdnellen , thanks for you immediate reply, 
the released jar file 190725.01 version is OK when generate verilog file.