Juniper/open-register-design-tool #77: ORDT coverpoints in RTL does not rename the resets to the system reset

neenuprince, 2019-11-14:
When the global reset is used it renames the reset signal to global_rst_n but it does not rename the reset signal for the coverpoints. The reset signal for the coverpoints is not an input and is unconnected.
example even if the system reset is named "syst_rst_n" the coverpoints in *jrdl_logic.sv uses "reset"  as below 
h2l_counter_rf_cnt_incrbyone_incrval_incr: coverpoint h2l_counter_rf_cnt_incrbyone_incrval_incr iff(!reset);

Each time it gets regenerated , we have to edit them manually.

sdnellen, 2019-11-17:
Kamino cloned this issue to [sdnellen/open-register-design-tool](https://github.com/sdnellen/open-register-design-tool/issues/16)

sdnellen, 2019-11-21:
Kamino cloned this issue to [sdnellen/open-register-design-tool](https://github.com/sdnellen/open-register-design-tool/issues/17)

sdnellen, 2019-11-23:
fix committed on sdnellen fork

neenuprince, 2019-12-18:
Confirmed that the fix works

closed 2019-12-18 by neenuprince