Juniper/open-register-design-tool #78: External Reset signals Causing RTL generation to fail

neenuprince, 2019-12-19:
Hi, Recently we noticed that the external reset registers, ( registers that gets reset using an external reset signal) that has resetsignal attribute causes an Error and hence cannot generate RTL.

We had generated the RTL with these resetsignal attribute earlier, but even older releases seems to fail. the error we get is as below.
*** ERROR ***: Unable to resolve rhs assignment reference test_rst_sig
Below I have mentioned the Reg defns and addr map.
Register Definition
==============
    signal test_rst_sig {  
       async;
    } test_rst_sig;

regfile other_regs_defns_rf {
 reg attr_resetsignal_t {
        name = "Reg resetsignal";
        desc = "A register with resetsignal to verify resetsignal works correctly";
            default hw = r;                //Register default hw field type to read/write
            default sw = rw;                //Register default sw field type to read/write 
        field {
            name = "rstsignal check";
            desc = "Verifying that rstsignal works";
            resetsignal = test_rst_sig;   //signal used to reset field's register and hardware logic
            fieldwidth = 32;
            reset = 32'h0;
        } fieldrstextsig;
    };
 attr_resetsignal_t    attr_rstsignal  
}; 

Top RDL address map 
================
addrmap ordtreg_regs {
 test_rst_sig    test_rst_sig;
other_regs_defns_rf         other_rf            @ 0x000100;
other_rf.attr_rstsignal.fieldrstextsig->resetsignal = test_rst_sig;

};

sdnellen, 2019-12-22:
Kamino cloned this issue to [sdnellen/open-register-design-tool](https://github.com/sdnellen/open-register-design-tool/issues/18)

sdnellen, 2019-12-23:
See sdnellen fork update - pls submit issues there going forward.