Juniper/open-register-design-tool #84: Control bit-width of address bus in addrmap

tenaliram, 2020-08-16:
I have 4 registers in my addrmap, but want to reserve 16 addresses. When I generate the address map verilog file, the h2d_pio_dec_address is only 2 bits (since I have only instantiated 4 registers). How can I force the h2d_pio_dec_address to use 4 bits?

ebertland, 2020-08-16:
You can create a read-only register at the bottom of your range with constant value of 0 and reset value of 0. This will synthesize away, but you will have the address width that you want in the RTL. I can post an example if this is not clear.

tenaliram, 2020-08-16:
Yes, please post an example of constant value=0 and reset value=0. 

I created a register with (SW=r), but that is still creating ports on HW side. So, I had to create a "HW=w, SW=r" register. 

One issue with creating a dummy register is that it will appear in the register documentation, if we use the RDL file to create the register documentation. I have multiple address maps which need this dummy register. If there is no other way to specify the address space for an address map, then I will have to use this method, but would like to know if there is an alternate method. 

Thanks for the help and quick response.