Juniper/open-register-design-tool #94: Solving decoding congestion

ehudeliaz, 2024-09-30:
Hi,

I have a unit with many registers, which leads to a big routing issue, so I wish to perform the decoding in multiple levels.
Basically I have a bunch of registers which repeat themselves, and I would want to have a separate decoding for each such group.
To do so, I have defined a group of the repeatable registers in an addrmap, and used the "generate child addrmap" feature.
The RTL is generated exactly as needed (though it needlessly generate a separate file per addrmap instance).
My only issue with this solution, is that I can't have an array of addrmaps, as the tool gives an error message:
 "Replicated address maps are not allowed"
So, I need to write multiple instances of the addrmap, leading to C files without structs, which makes the SW code very cumbersome.

What is the reason for preventing array of addrmaps?
Do you have a better solution for this kind of issue?

Thanks!

sdnellen, 2024-10-07:
Repeated addrmaps caused some corner cases that were not resolved - repeated addrmaps were not a common case and workaround is to wrap repeated regfiles in an addrmap.  This adds a level of hierarchy which is annoying, but generally workable.

ehudeliaz, 2024-10-09:
Hi,

Thank you for the answer. 
But, as I understand, this solution doesn't refer to the basic congestion issue.
I want each repeated group of registers to have their own address decoding stage in hardware.
As I understand, to get a separate address decoding stage I must use addrmap with "generate child addrmap" (and as we know, addrmap can't be repeated).

Can I get a separate address decoding stage using regfile?

sdnellen, 2024-10-11:
OK, misunderstood your use case - think you are looking for N interfaces from the base decoder to N identical child decoder/reg groups?  If so, could do something like:

regfile {
       my_addrmap norep_map;  // creates 4 separate interfaces from base decoder
} rep_wrapper_rf[4] @0x300;